ul-gh/PiPyADC #4: Python3 Operation

parallellogic-, 2017-09-30:
This code natively runs in Python2, but I require a Python3 environment for my project.  Trying to run the code as-is produces segmentation faults with wiringpi.wiringPiSPIDataRW() and with wiringpi.delayMicroseconds().  The following changes allowed me to resolve those issues and run with Python3:

WAS: wiringpi.wiringPiSPIDataRW(self.SPI_CHANNEL,"%s" % chr(mybyte&0xFF))
IS: wp.wiringPiSPIDataRW(self.SPI_CHANNEL, bytes([mybyte&0xFF]))
-
WAS: MISObyte = wp.wiringPiSPIDataRW(self.SPI_CHANNEL, chr(0xFF))
IS: MISObyte = wp.wiringPiSPIDataRW(self.SPI_CHANNEL, bytes([0xFF]))
-
WAS: self._DATA_TIMEOUT_US = 1 + (50*1000000)/conf.CLKIN_FREQUENCY
WAS: self._SYNC_TIMEOUT_US = 1 + (24*1000000)/conf.CLKIN_FREQUENCY
WAS: self._CS_TIMEOUT_US   = 1 + (8*1000000)/conf.CLKIN_FREQUENCY
WAS: self._T_11_TIMEOUT_US   = 1 + (4*1000000)/conf.CLKIN_FREQUENCY
IS: self._DATA_TIMEOUT_US = int(1 + (50*1000000)/conf.CLKIN_FREQUENCY)
IS: self._SYNC_TIMEOUT_US = int(1 + (24*1000000)/conf.CLKIN_FREQUENCY)
IS: self._CS_TIMEOUT_US   = int(1 + (8*1000000)/conf.CLKIN_FREQUENCY)
IS: self._T_11_TIMEOUT_US   = int(1 + (4*1000000)/conf.CLKIN_FREQUENCY)

https://pastebin.com/CHnCLXii

ul-gh, 2017-10-01:
Thank you very much for your feedback!

I am on vacation until 4th of October - I will look into this afterwards - promise.

However, if you want you can also check out the "pigpio" branch of the software, which uses the pigpio library instead of wiringpi, except the wp.delayMicroseconds() function where I found wiringpi to be significantly more accurate for small delays than time.sleep(). It is quite possible that pigpio is better supported for python3, and anyways it is more full-featured and offers very accurate timer and io polling callback functions. Wiringpi still has a small performance-advantage on my hardware, which should not matter for the intendet application.

Best wishes,
Ulrich

ul-gh, 2017-12-22:
Hi again,

yes, I would agree that the classic Python 2 floor division / missing explicit integer cast is a bug, also it seems that Python 2.6 and later support a "bytes" dummy type for compatibility with Python 3 code.

==> I will add your fixes to the code when I get my hardware back for testing, either on the second week of January 2018 or sooner, if I can borrow a Raspi.

Regards,
Ulrich

kizniche, 2018-10-16:
Thanks for a great library. I, too, am restricted to the python 3 environment. However, @parallellogic-'s [comment](https://github.com/ul-gh/PiPyADC/issues/4#issue-261876225) worked to allow python 3 execution. I [forked this repo](https://github.com/kizniche/PiPyADC-py3) and added setup.py so I could use pip to use it as a dependency until this repo becomes python3-compatible.

ul-gh, 2019-02-14:
Hi, this issue should now be resolved via the latest commits. If there are any problems, please reopen this issue.

closed 2019-02-14 by ul-gh