ul-gh/PiPyADC #5: Unable to open SPI device: No such file or directory

BirdingPix, 2017-12-22:
I finally am asking for a bit of help since I cannot solve this myself. 

I get this error when running example.py (or benchmark.py):
Unable to open SPI device: No such file or directory

The error occurs when executing this line in pipyadc:
        fd = wp.wiringPiSPISetupMode(
                conf.SPI_CHANNEL, conf.SPI_FREQUENCY, conf.SPI_MODE)
 
[image]

The values are correct but unable to get past this error.  
Pls offer some guidance... THANK YOU!

I know the Waveshare High-Precision AD/DA hardware is running ok as their little demo program works (sudo ./ads1256_test).  but I need to run this within PYTHON environment.

thanks again for any help!  
Dave
Houston, TX

ul-gh, 2017-12-22:
Dear Dave,

this looks as if the Linux device file is missing - which could be a kernel configuration or kernel version issue.

Can you post the output of the following commands:

ls -l /dev/spi*
grep "^dt.*" /boot/config.txt
uname -a
cat /etc/os-release
lsmod

I do not have my Raspberry Pis with the ADC board at hand as I am on Christmas vacation for the next two weeks. Maybe I can borrow one from a friend for testing this. Maybe this is not even necessary.

Let us fix this before Christmas!

Best wishes,
Ulrich

BirdingPix, 2017-12-22:
Ulrich,
THANK YOU so much for fast reply.  I've continued working on it and just got it working!  Even though the other test program worked... and raspi-config said SPI was ON... I changed the /boot/config.txt file to uncomment  dtparam=spi=on andf also added dtoverlay=spi-bcm2835 

Not sure which fixed it, but now for the first time... your example.py is working so I can progress!

THANK YOU!   I'll keep you posted on my progress.
Dave