ul-gh/PiPyADC #6: differential input

sarbot, 2018-03-10:
first of all let me say good job! I've looked for a while for useful python code to use my AD/DA High-Precision Board, since the documentation for it is really bad as long as you are not familiar with C to understand the example code.

I didn't read the complete Class code to be honest.. just the example files. Am I doing right to build input pairs of two ADx-Inputs to get the differential voltage between two pins and remove the AINCOM-AGND jumper? For example(.py):

    DIFF1, DIFF2 = POS_AIN1|NEG_AIN2, POS_AIN3|NEG_AIN4
    CH_SEQUENCE = (DIFF1, DIFF2)

or do I need to setup anything else in the config/initiation of the ads object? I'm used to python, but have no clue about electric stuff and microcontrollers, thats why I ask... just to be sure ;).  Further can I use a simple AAA-Batterie (1.5V) or whatever (below 5V ofc)  to check if the variables and wirings are correct in my setup?

sarbot, 2018-03-10:
Nevermind as I was writing this it seems legit enough to myself to try it and it works. Thanks for this simplicity was too confused by trying to use the bcm2835 C library ;)

closed 2018-03-10 by sarbot

ul-gh, 2018-03-10:
Hi,

yes, although you do not even need to remove the AINCOM-AGND jumper.

When the jumper is set, the separate AINCOM pin of the ADC is connected to the screw-type terminals of the board, but due to layout imperfections, this is influenced by on-board stray currents causing offset drift. With the jumper removed, you can connect the AINCOM pin using a connector lead to the jumper, ruling out stray current influence (but chances are you could get a noise issue because of missing filter capacitors).

sarbot, 2018-03-10:
Good to know I'll check how much that matter in practical use. Thanks for your answer.