ul-gh/PiPyADC #8: Segmentation fault on Raspberry pi 3 B+

adn05, 2018-06-19:
Hi, firstly it is a good job you have done here. 
I'm trying to execute the example.py on a Raspberry Pi 3 B+, but I got a Segmentation fault.
When I looked closer, I noticed that it is during the execution of **_send_byte()** that it fails. 
I don't know if you've ever seen this or if it is at wiringpi level the problem is.

ul-gh, 2018-06-19:
Hi, yes, there is likely something wrong with your wiringpi installation. Or you have a permissions/access problem, normally the code must be run as root.

If wiringpi does not work for you, you can also use pigpio. This is the respective branch:
https://github.com/ul-gh/PiPyADC/tree/pigpio

HTH, Ulrich

adn05, 2018-06-20:
Hi, the pigpio branch works fine for me. Thanks.
I got a TypeError in delayMicroseconds method of WiringPi, that I solved by casting the delays to int. I'm using python 3.5.
`TypeError: in method 'delayMicroseconds', argument 1 of type 'unsigned int'`

I will make a pull request for it.
Narcisse

ul-gh, 2018-06-21:
Hi, thanks again for your error report, please see the latest commit regarding the type error.
Regarding the onboard DAC, I am afraid, no, I did not do any work with it yet.

closed 2018-06-21 by adn05