ul-gh/PiPyADC #10: Reading two Waveshare ADS1256 board interfaced to the Raspberry Pi 3

glatzii, 2018-06-25:
Hello Ulrich Lukas!

Thank you for you work, it helped me a lot! I am reading out several sensor values via the waveshare board on the RaspPi and your code works like a charm.

However, I need more than 8 channels so I wanted to stack 2 boards. Easier said than done because I can't distinguish between 2 boards, they do affect each other. I did what was written in your example 2 but when I:
    print ads.chip_ID
    print ads2.chip_ID
I get the same numbers and I cannot readout different values. I provided 2 different config files but do I have to change something in there? I think I miss only a small thing.

Kind regards,
Thomas
[ADS1256.zip](https://github.com/ul-gh/PiPyADC/files/2133330/ADS1256.zip)

ul-gh, 2018-06-25:
Hi Thomas,

this is a hardware design limitation of the cheap Waveshare boards which  do not have individual chip select lines. I.e. the stock PCB layout does not allow stacking of the boards to address more than one ADS1256.

In order to address more of these SPI slaves, you have to re-route the chip select pin of all additional ADCs to an individual, free GPIO pin and make the modifications in the configuration. Even better would be to use the hardware chip select lines of the Raspberry Pi SPI output, but for some reason unknown to me the designers of the Waveshare PCB used a normal GPIO. What that does allow you to do however is to use more than two boards.

HTH,
Ulrich

ul-gh, 2018-06-25:
adding to this issue, the DRDY pin also needs re-wiring for multiple Waveshare boards to work.

If you look at the pinning of the Waveshare board, pins 18 (GPIO 24) and 22 (GPIO 25) are listed as not connected in the table.
> https://www.waveshare.com/wiki/High-Precision_AD/DA_Board

I.e. if you place a wire connection from pin 11 to pin 18 and from pin 15 to pin 22 on the second ADC board and then cut away or desolder pin 11 and pin 15, you should be good to go.

Please let me know if this works for you. Regards, Ulrich

ul-gh, 2018-06-25:
BTW, the "Chip ID" is identical for all chips AFAIK. These are discerned by their chip select lines.

glatzii, 2018-06-28:
Good Morning Ulrich!

Thank you so much for your help and advise! It works perfectly, I am able to read out the two boards which are stacked. You saved me a lot of time.

Best wishes,
Thomas

closed 2018-06-28 by glatzii