ul-gh/PiPyADC #11: Reading voltages at the screw input terminals

leloup314, 2018-07-19:
Hello,
first of all, great work on this. It works quite nicely out of the box. Since I am a newbie when it comes to hardware programming / RPi I have some questions regarding on how to use this code to read out voltages at the screw terminals:

I want to read out 8 single-ended voltages on the screw terminals AD0 to AD7 compared to ground. Somehow, when I run your `example.py` I get the Poti_CH0 and LDR_CH1 voltages (I connected the respective jumpers on the right side: VCC=5V=VREF, ADJ on AD0, LDR on AD1, AINCOM on AGND) but then the other channels  (AIN2, AIN3,..) show some random values.  When I now connect let's say 0.8 V signal to screw terminal AD2 I'm expecting to see AIN2 to go to 0.8 V but it doesn't. My question is:
How do I define the channels at the screw terminals AD0 to AD7 to be read out single-endedly with respect to ground?

I'm sorry if this is straight forward and I'm just to blind to see.
Cheers

ul-gh, 2018-07-19:
Hi, with the example configuration, this should work out of the box:
EXT2, EXT3, EXT4 = POS_AIN2|NEG_AINCOM, POS_AIN3|NEG_AINCOM, POS_AIN4|NEG_AINCOM

Did you connect the GND wire of the 0.8V voltage source to AINCOM/AGND?

leloup314, 2018-07-20:
Hi,
  
thank you for the reply. I did not do that because I assumed since    the RPi GND jumper was set it would compare the 0.8 V to (some) ground. Should I connect the 0.8 volt s ground to the GND screw terminal then? You might as well label this issue as `question`. In will try everything on Monday! Have a nice weekend!
  

  
Cheers

leloup314, 2018-07-25:
Hi,
everything works fine now. In my hurry I didn't connect ground at all so as you pointed out that was my mistake. Thanks!

closed 2018-07-25 by leloup314