ul-gh/PiPyADC #12: ADS not initializing...

jwebster8, 2018-08-07:
I have a new RPi3 with a new waveshare High-Precision AD/DA board.

I installed your code, but when I run the examples, the board doesn't seem to initialize...I put a couple of print statements in to see where it was making it too..

```
def do_measurement():
    print('doing measurement\n')
    ### STEP 1: Initialise ADC object:
    ads = ADS1256()
    print('ADS1256 initialized\n')
    ### STEP 2: Gain and offset self-calibration:
    ads.cal_self()
    print('ADS1256 self-calibrated\n')
    while True:
        ### STEP 3: Get data:
        raw_channels = ads.read_sequence(CH_SEQUENCE)
        voltages     = [i * ads.v_per_digit for i in raw_channels]

        ### STEP 4: DONE. Have fun!
        nice_output(raw_channels, voltages)
```

and it outputs...

> PiPyADC: Example file for class ADS1256 in module pipyadc:
> 
> ADS1256 cycling through eight input channels.
> 
> Hardware: Waveshare ADS1256 board interfaced to the Raspberry Pi 3
>  
> Ulrich Lukas 2017-03-10
> 
> 
> Press CTRL-C to exit.
> doing measurement

but nothing else.  So Its failing at    `ads = ADS1256()`

Am I forgetting something?  It runs with some C code, so I don't think it's a hardware problem.

Thanks

ul-gh, 2018-08-07:
Hi,

what is the output if you run:
```
import wiringpi as wp
wp.wiringPiSetupPhys()
```
?

jwebster8, 2018-08-07:
> 0

ul-gh, 2018-08-07:
Thanks for your answer - that basically only means that you have the wiringpi libraries installed and this should be working.

What you could do is run the functions from the __init__ function of class ADS1256 (file pipyadc.py) one after the other or debug from this point:
https://github.com/ul-gh/PiPyADC/blob/6b314561c47e3cada06c24e0dd35286d6251e6a5/pipyadc.py#L267

I can't help you more than this without further information, I am afraid.

ul-gh, 2018-08-07:
Hi, not sure, (I've had the second beer for today, it's past midnight here and I also don't have access to the hardware) but this could be a WirintPi issue. You can try out the PiGPIO branch if you want: https://github.com/ul-gh/PiPyADC/tree/pigpio. If you want, I can look into this later, but today - not a chance.. Best wishes, Ulrich

jwebster8, 2018-08-07:
we can take this up again tomorrow.  Enjoy the beer

jwebster8, 2018-08-08:
Yep, it worked with the pigpio branch

ul-gh, 2019-02-14:
This could be related to the Python 3 issue #18, which is fixed via the latest commits. If there are still problems, please reopen this issue.

closed 2019-02-14 by ul-gh