ul-gh/PiPyADC #13: Lower Voltage value readings when increasing samping rate

gabrielecostante, 2018-08-08:
Hi,

I successfully installed your code and I am able to run your example.py script without any problem (I only needed to switch to the pigpio branch, since the wiringpi gave me segmentation fault issues). 

However, when I tried to increase the sampling rate by inserting the line
`ads2.drate = DRATE_100`
in the example.py file, I noticed that the voltage value readings are lower (and wrong) with respect to the ones given by the original example with the default sampling rate.
Am I missing something? Is there something that needs to be done when increasing the drate parameter?

Thanks for your help

ul-gh, 2018-08-08:
Hi,

this can happen if the analog inputs are subjected to noise voltages, very high source impedances or especially if they are left unconnected and floating. Otherwise, this is an error.

What do you have connected to the inputs?

gabrielecostante, 2018-08-09:
Hi, 

this happens both when using the built-in potenziometer for testing (I am using the Raspberry Pi AD/DA Expansion board with the ADS1256 chip) and by connecting an external power supply. The voltage measured, in both cases, seems to be nearly halved.

gabrielecostante, 2018-08-10:
Yes, the jumper is connecting AGND to AINCOM.

leloup314, 2018-09-13:
I have also seen this for voltages which come close to the 5 V region when using the `ADS1256_default_config.py` without any changes. For example for 4.7 V input the ADC reads *3.9* V, the voltage even seems to decrease over time for constant input voltage. For me this was only the case for high voltages (> 4 V) in the lowest resolution e.g. `gain_flags = GAIN_1`! For other internal gain I did not see this. Since I need to use th full +-5V range I tried fixing the issue. What worked for me was to set the status from `BUFFER_ENABLE` to `AUTOCAL_ENABLE`.

ul-gh, 2018-09-15:
Many thanks to Pascal Wolf, this seems to be the crucial bit of information:

Indeed, the ADS1256 has a limited input voltage range of 0V...3.0V for a 5V supply when the input buffer is enabled:
[image]

You must disable the BUFFER_ENABLE flag for voltages above 3.0 V to work reliably.

closed 2018-09-15 by ul-gh