ul-gh/PiPyADC #14: Full scale is not +/-5V

Shlior, 2018-08-12:
Hi,

I'm evaluating my setup. Injecting sine wave at 1Hz +/-5 volts to channel A0.

Configuration:
The gain setting (PGA) is 1.
50 SPS.
AINCOM is connected to AGND.
VREF 2.5V.

I'm expecting +/-5V full scale.. I get somewhat around +~4V and ~-0.15V.

What can be the cause for such behaviour?
[image]

A correct behaviour example +/-0.13V sine:
[image]

Thanks,
Shlomi

ul-gh, 2018-08-12:
Hi,

what i see from your graphs is clipping.

In this case, this has two reasons:

A)

The ADS1256 has typical CMOS inputs i.e. the input common-mode voltage
range is GND...VDD and the internal gate electrodes are ESD-protected by
silicon diode junctions.

The inputs cannot go significantly below the GND or above VDD potential.

If you want to measure AC voltages, you have to add a DC offset. The
easiest method is a high-value resistor to the 2.5V REF voltage and an
AC coupling capacitor in series with the AC source. You can also
resistively add the DC offset using two resistors.

If you do this, of course you have to make sure that the automatic input
offset-voltage calibration happens at the right moment with actually
zero volts applied.

B)

If you enable the ADS1256 input buffer amplifier (this is the default
configuration with the BUFEN flag), the analog input voltage range is
further limited to a guaranteed "AVDD - 2.0V" according to page three of
the datasheet.

You can then simply disable the integrated buffer amplifier and do
without. If you have a high-impedance voltage source, you can add an
external rail-to-rail operational amplifier as a buffer.

You can also just resistively divide any input voltage range to the
range specified in the datasheet.

Shlior, 2018-08-13:
Hi,

Thanks, I see now that the inputs are ESD protected, meaning the analog inputs cannot go below GND (minus diodes forward voltage).

Thanks,
Shlomi

closed 2018-08-13 by Shlior