ul-gh/PiPyADC #17: Initialization causes software crash when board is not connected

kizniche, 2018-10-16:
When initializing the ADS1256 class (```adc = ADS1256()```), my software crashes without an error.

I've briefly attempted to diagnose this issue, however I don't receive any errors when this happens, making it difficult to debug. If you have any recommendations for what might be the issue or how I can check if the board is connected before attempting initialization, it would be most appreciated.

kizniche, 2018-10-16:
As a note, in case it's relevant, I've performed a [minor edit to the code](https://github.com/ul-gh/PiPyADC/issues/4#issuecomment-430400563) for python 3 compatibility.

ul-gh, 2022-07-01:
Hi,

with version 2 of this code, I added proper error handling, please have a look and let me know if there are still any issues.

Best wishes,
Ulrich

closed 2022-07-01 by ul-gh