ul-gh/PiPyADC #20: Clarification regarding the selection of ADC Channels

vandan-revanur, 2019-09-10:
Could you please elaborate as to why the the version implemented uses software/GPIO setting of the chip select signal instead of using the SPI hardware ?

I would like to know if GPIO setting using Chip select pins has some kind of advantage over using the SPI hardware.

Thanks in advance.

ul-gh, 2019-09-10:
Hi,

the software chip select is because of the "Waveshare" brand hardware boards where the ADC CS pin is simply not connected to the right pin on the Raspberry Pi board.

In theory, this makes sense if there are more GPIOs than specialised chip select outputs so you can connect more devices by using GPIOs. In the case of the Waveshare board, I suspect they just mixed up the pinning and this is only a workaround.

ul-gh, 2022-07-01:
Hi,

please look at #21 description - the ADS125x devices are indeed incompatible with the common "hardware chip select" modes so this explains why the Waveshare board implemented this from the start.

Best,
Ulrich

closed 2022-07-01 by ul-gh