ul-gh/PiPyADC #21: Hardware Control of CS Pin

jliriano2000, 2019-12-21:
Hi! 
I saw there was a ‘FIXME’ in the default config file around hardware control of the CS pin. I reworked the ADC1256 Pi board to use SPI CE1 instead of the GPIO based CE.  I updated the config file with the new pin and it works except when I raspi-GPIO get, I see the CE1 pin is GPIO.  I see it’s reassigned  in PiPyADC.py. (I have it in hw SPI mode before.)

But when I assign None to the Ce pin in the config file, I get nothing but zeros when reading. 

Why is this? Does wiring pi not support HW CS? What would it take to fix this ?

ul-gh, 2020-02-22:
Hi,

one more better late then never reply:

WiringPi supports the chip select hardware line.
You should only have to set CS_PIN to None in the configuration and then it should work.

If not, then there is a bug that I would have to work on.

The ADS1256 chip select pin is active low, which is also the only option for WiringPi and this should work. Except if you have an extra hardware inverter on your board.

If you want, you can send me the reworked board for testing..

jliriano2000, 2020-02-22:
Hey there!

So setting the CS_PIN to "None" did not work for my implementation. I
believe it is an issue with wiringPi itself. It did not seem to have fine
control over the hardware when it came to the CS pin. When used in GPIO
mode, the SPI bus worked perfectly.

I got better results with SPIDEV 3.4. I was going to see if I could replace
wiringPi with SPIDEV in the pipyadc source.

ul-gh, 2022-07-01:
Hi,

I can now report that the reason why hardware chip select control does not work with the ADS125x devices and using the pigpio or wiringpi library is because the ADS125x hardware devices need a delay between sending a command and reading back the data over the SPI bus, while expecting the chip select line to stay active during the delay.

This is incompatible with the common hardware-based SPI chip select modes as these set the chip select line inactive between each data telegram.

This is why in version 2, I removed the comments about hardware chip select options.
It still works fine with "bit banging" mode, plus you can also tie the CS pin permanently to GND if you only have one ADS125x chip on the bus.

Best wishes,
Ulrich