ul-gh/PiPyADC #25: ADS1255 Read Id fail

Jerzeek, 2020-11-10:
Hi All :) 
I made a [board](https://easyeda.com/jerzeek/nir-spectroscope) with the ADS1255
I have changed the pins in the default config file, but I can get successful readings or a good chip ID
What can I do to make it work?

ul-gh, 2020-11-10:
Dear Jerzeek,

Am 10.11.20 um 09:55 schrieb Jerzeek:
> I made a board <https://easyeda.com/jerzeek/nir-spectroscope> with the
> ADS1255
> I have changed the pins in the default config file, but I can get
> successful readings or a good chip ID
> What can I do to make it work?
This looks very interesting, I guess this spectroscope is used for
sorting plastics?

However, as for your question, there are a lot of possible reasons that
there is no communication to the ADS1255. If the hardware is supplied
with the correct voltages, the SPI pinning is correct, the SPI bus is
the correct one (on the Pi4, there are SPI0...SPI6), there could also be
an isue with the WiringPi version that I don't know about.

Did you hot-plug the board  onto the Pi? In this case, communication is
not set up correctly, the board must be plugged in before booting the Pi.

To be honest, I have only tried the ADS1256 chip since the ADS1255 is
just less channels in comparison with the ADS1256 if I read the
datasheet correctly.

Since we are on lock-down, I could have a look at your hardware if you
send me a board, I will send it back after a week or two.

Jerzeek, 2020-11-16:
Hi Ul-gh!
Wauw, sharp eye, indeed, a spectroscope for plastic identification :D 

I have played with the code and somehow got it to work! I wrote my own, most basic, example to read chip ID and in the end got it to work, I still have some small timing issues (maybe because I'm using a 8mhz clock) if I fix these issues then I will post the exact changes I made to get it to work

Thanks again for your reply