ul-gh/PiPyADC #26: Random voltage peaks

nchazarra, 2020-12-10:
Hi,

I'm usind PiPyADC to sample at 500 sps three differential channels (I'm getting geophone reads) at gain one. Some time, random peaks around plus/minus 65000 counts appear while background noise levels are around ~100 counts. 

Does anybody have the same issues? Thanks in advance.

ul-gh, 2022-07-01:
Hi,

admitted this might be irrelevant for your project now, but please have a look at the new version and/or if you found a solution.

I can not reproduce the random outlier values, but leaving this open in case there is an update from your side.

nchazarra, 2022-08-25:
HI @ul-gh,

I can't test it right now, but I'll be testing it, of course! Many thanks. If I find that this issue has been resolved, I'll tell you.