ul-gh/PiPyADC #27: SPS 3000 Python Problem

Kakalos14, 2021-01-15:
Happy new year, 

I hope you are all healthy. I'm trying to get a 1sec sample with 30000SPS from ADS1256 component with Python and write it to a file, based on this example: https://github.com/ul-gh/PiPyADC/blob/master/examples/example_2.py, but i cannot reach the 30000SPS data rate, not even a data rate near 1000SPS. Is it due to read_sequence function, due to Python issues or something else? Could you please suggest a solution in order to reach a data rate at or near 30000 SPS and write it to a file at real-time(or almost real-time)?

Thank you in advance!

ul-gh, 2021-01-17:
Hi,

when using this driver, we receive single bytes and do bit-banging of hardware signals - all from code running in the Python interpreter.

This simply does not allow the 30 kHz sample rate.
Depending on your hardware and system, you can get more than 1000 SPS, however what this 24-Bit ADC excels at is doing much slower sample rates, i.e. high-precision applications. For this you get good results.

For really high sample rates, you can get this using buffered reads of continuously sampled values - but then again you need an ADC device which has the necessary hardware support, e.g. a built-in sequencer if you have more than one channel.

My suggestion is look for one of the many ADCs which have native Linux kernel driver support. There are some high-bandwidth devices with good support.

hayat1993, 2021-02-04:
Hello 
 I have the same problem , can u help me if u find solution?

ul-gh, 2021-02-05:
Dear hayat,

as it says above, you can not get more than a few hundred or a few thousand samples per second when reading single bytes using a python interpreter running in a user task under a multitasking operating system.

No. This is absolutely not possible.

For that to work, you would have to write a kernel driver likely having a part of it running in interrupt context even.
Or use a hardware buffer.

Otherwise, when you do high-resolution acquisition, the delta-sigma converter hardware uses a CIC filter internally which is effectively nothing else than an optimized moving average, and this is what yields the high precision of the delta sigma converter.

If you need noise rejection of 50 or 60 Hz line frequency, you are stuck with a slow acquisition rate anyways.

Closing this as won't fix..

closed 2021-02-05 by ul-gh