ul-gh/PiPyADC #30: doesn't work on bullseye?

cmd2001, 2021-08-30:
My hardware is raspberry pi cm4 with waveshare CM4-IO-BASE-A, and I'm using waveshare High-Precision AD/DA Board.

My software version is Python 3.9.2 on Debian 11 bullseye, I got this system by simply upgrading from official Debian Buster distribution.

When I tried to get voltage input from the ads1256 extension board by executing the example.py, it always returns 0v.

However, when I tried waveshare's example, it got the correct value.

Here is the output of pypiadc example:

```
PiPyADC: Example file for class ADS1256 in module pipyadc:
ADS1256 cycling through eight input channels.
Hardware: Waveshare ADS1256 board interfaced to the Raspberry Pi 3
 
Ulrich Lukas 2017-03-10

Press CTRL-C to exit.

These are the raw sample values for the channels:
Poti_CH0,  LDR_CH1,     AIN2,     AIN3,     AIN4,     AIN7, Poti NEG, Short 0V
       0,        0,        0,        0,        0,        0,        0,        0
These are the sample values converted to voltage in V for the channels:
Poti_CH0,  LDR_CH1,     AIN2,     AIN3,     AIN4,     AIN7, Poti NEG, Short 0V
   0.000,    0.000,    0.000,    0.000,    0.000,    0.000,    0.000,    0.000

```

And the following is the output of waveshare's example:

```
ID Read success  
0 ADC = 4.227387
1 ADC = 4.354094
2 ADC = 1.147323
3 ADC = 1.146529
4 ADC = 1.147040
5 ADC = 1.145547
6 ADC = 1.014713
7 ADC = 1.013647
```

Is there anything wrong with pypiadc or python 3.9?

ul-gh, 2021-08-30:
Hi,

I'm afraid I can't test your setup right now.

You could first test if your SPI communication is working OK, e.g. if
you read the chip_ID property, this should be a number "3".

If this does not work, and seeing that WiringPi is now obsolete, please
try out the "pigpio" branch of this repo:

https://github.com/ul-gh/PiPyADC/tree/pigpio

I am working on a completely new, DMA based high-speed driver for this
chip, inspired by:
https://iosoft.blog/2020/05/25/raspberry-pi-dma-programming/
https://iosoft.blog/2020/06/11/fast-data-capture-raspberry-pi/

However, I cannot promise any results soon.

ul-gh, 2022-07-01:
Dear Amagi,

please have a look at the new version 2.0.1 if this is still relevant for you.

The new version uses pgpio instead of wiringpi, please look at the - updated - install instructions.

I just now tested with python 3.9.2 on Raspberry Pi OS current version (Debian 11 "Bullseye") on a completely fresh install on a Raspberry Pi 2B and also on a Raspberry Pi 3B, and it works for me.

Please feel free to open a new issue if there are any problems.

closed 2022-07-01 by ul-gh